Call-for-Code/Prometeo-Pyrrha #2: Recruit IBM Service Corps team for roles

krook, 2019-12-03:
Hardware engineer
Firmware engineer
Developer
Design/User Experience/Branding
Architect/advisor/advocate/IBM Cloud expert
DevOps/Infrastructure specialist
Data scientist/Analyst
Health/domain expert
Tester/QA
Digital content creator
Communications/marketing/ecosystem liaison
Logistics/operations
Project manager/scrum master
Legal/regulations/security

developersteve, 2019-12-03:
Two things i wanted to make note of (after spending time with the crew in Australia recently), 

There totally needs to be a catm1/narrowband version or option... 

Also going down the open source community project route id also have a community manager as part of this too (or potentially part of the comms role)

krook, 2019-12-03:
Thanks for the input @developersteve.

I'm not sure we captured the cacatm1/narrowband issue just yet. I'll make sure that gets in.

Agree on the community manager piece, agree. Here are the roles-to-tasks being brought into GitHub.

[image]

[image]

krook, 2020-01-21:
Team has been recruited. Once they accept their roles we will assign them tasks to help virtually ahead of the February field test and prepare for deployment in April.

closed 2020-01-21 by krook