Call-for-Code/Prometeo-Pyrrha #41: Data scientist/Analyst: Determine how to customize alerts for each individual firefighter profile

krook, 2019-12-03:


JSegrave-IBM, 2020-06-24:
Note: in order to work out the tech (algos / ML) behind the alerts, we're going to need to do some of the storyboarding described in [Issue 89](https://github.com/Code-and-Response/Prometeo/issues/89)

e.g. (illustrative example)
In the command center, an alert goes red for Alfonso. What happens next?
* Does Alfonso's response depend on knowing what the underlying condition, or conditions were? e.g. will he react differently if
** a sensor just saw a toxic level of NO2 for a few seconds and then stopped
**  a firefighter is experiencing a combination of {high temp + CO}?
**  a firefighter has been exposed to too much NO2 cumulatively over the last 15 minutes?

Labelling data is one of the big costs in machine learning and the 'type' of explainability required determines how we do the labels (as well as how feedback is gathered at runtime.)  e.g. red / yellow /green is simple, but is it sufficiently-explained to enable correct follow-up actions? When context/explanation is essential, we often choose to use machine learning to learn the *explanations* (like _'critical exposure to NO2 over 15 mins'_) rather than the _decisions_ (_'red: get the firefighter out'_).  Need to know this up-front, as labelling 100s/1000s of examples is expensive is usually not cost-effective to repeat / fix.

krook, 2020-09-08:
I think we'll need to include this in a set of assumptions and future recommendations based on the research we've done.