Call-for-Code/Prometeo-Pyrrha #53: Logistics/operations: Prepare scope for February field test

krook, 2019-12-03:


svaleroc, 2019-12-05:
Tentative Monday, Februry 17
Number of firefighters for first field test (Friends&Family): 10
Previous test on the field with Joan Herrera
The final date will depend on weather conditions and final calendar for prescribed burns

developersteve, 2019-12-05:
Is it possible to get some numbers on mobile signal/wifi close to fire, I’m
curious to see how it affects the signal

krook, 2019-12-10:
February 3rd-7th is general GRAF training.
February 10th-14th is the prescribed burn testing.

svaleroc, 2020-01-03:
February 10th-13th is the first prescribed burn testing. GRAF Firefighters has blocked their agendas theses 3 days to be available for the burns.
March 9th-11th could be the second prescribed burn testing. GRAF Department has already scheduled theses 3 days in March for the burns.

svaleroc, 2020-01-03:
The exact locations for the prescribed burns are not yet known but will be in the territory of Catalonia. The maximum distances will be an hour or hour and a half by car from Barcelona.

joan-herrera, 2020-01-15:
Under revision location exactly prescribed burns February 10 th-13th
https://goo.gl/maps/QuEWiuZ1F668SWEL8

krook, 2020-02-09:
Looks like everything is ready to go and in place for February 10, the start of the test.

closed 2020-02-09 by krook