Call-for-Code/Prometeo-Pyrrha #58: Architect/Advisor: Investigate integration between Prometeo and Copernico

krook, 2019-12-03:


mrodrise, 2019-12-04:
The idea is to upload data from Prometeo to the Copernicus database.

This Copernicus database has an API in order to download the info.

You can see an example here:
https://cds.climate.copernicus.eu/api-how-to

Anyway, the idea Is to upload data to this Copernicus System but we need to understand the objective and the requirements. So, we have to ask to Marc Castellnou

We have a meeting the 16th of December with the Copernicus Team in order to get knowledge about the integration with this system

mrodrise, 2019-12-23:
We're still waiting for the meeting, the GRAF firefighters are trying to schedule it. Maybe after New Year's Eve.

closed 2020-05-05 by svaleroc