Call-for-Code/Prometeo-Pyrrha #69: Prepare Field Testing in several sites with focal points from different locations (California, Australia, Argentina, GRAF Barcelona)

svaleroc, 2020-05-05:


svaleroc, 2020-06-17:
Working on the subject to create several firefighters' teams in different locations. Tentative Barcelona, Andalucía, Australia, California and Argentina. 5 sites with around 5-10 firefighters in ach location. GRAF coordinator Joan Herrera and asking for candidates through Pau Costa Foundation: https://mailchi.mp/880b7e07eb6d/opportunity-40would-you-like-to-be-the-first-one-worldwide-to-use-prometeo-field-test

krook, 2020-06-18:
We'll also need to connect with the folks in Australia via @developersteve and @rampadc 

@lidderupk and @samantadabis let's connect about the IIIF, who Mike Sefanov introduced us to at the Design Thinking workshop.

@SolSantillan do you have local contacts in Latin America? I know Pedro Cruz has a contact in Puerto Rico he wanted to engage.

krook, 2020-09-08:
@svaleroc is managing the schedule and contacts.