Call-for-Code/Prometeo-Pyrrha #72: Migration Cloud account

svaleroc, 2020-05-06:


krook, 2020-05-06:
Derek to work with Marco on this

krook, 2020-05-19:
How is this coming along @derekteay ?

derekteay, 2020-05-19:
Just got a note from IBM today, it says:

Hello Josep,

Thank you for contacting us. I reviewed the account account@prometeoplatform.com and found it's still a trial account. We can't move resource from your account to trial account. 
Please contact IBM sales team to set up an enterprise for account@prometeoplatform.com which will  allow you keep your account and also consuming the credits from the enterprise account. 

Please see more information about enterprise account from below link. 

https://mediacenter.ibm.com/media/t/1_hi08z2zm
https://cloud.ibm.com/docs/account?topic=account-enterprise-tutorial

Kind Regards,
IBM cloud support

I'm going to reach out to Ryan Johnson about this.

derekteay, 2020-05-20:
Ryan Johnson doesn't handle migration and doesn't have a suggested person on who to reach out to. 

After review, I believe this is the best course of action - https://cloud.ibm.com/docs/account?topic=account-enterprise-tutorial#create_enterprise_tutorial

I can do this or Marco can do this, since I don't have the credentials to the account at the moment. 

@mrodrise - do you want to perform this step? We can then update the case and get this moving, hopefully before the June 1st date.

krook, 2020-06-15:
We decided to recreate the resources in a new IBM Cloud account for Prometeo Platform.

The account has been created, and @mrodrise owns adding folks to it to help stand up a new version of the platform on Kubernetes with @lidderupk and @krook. 

This is the proposed set of new consolidated services:
```
- Kubernetes cluster
    - WS Server Dockerfile
    - Find out if Node-RED runs on the cluster
    - Rules engine (if needed, can wrap the machine learning model)
    - Dashboard (Python dashboard)
- IoT Platform
- Watson Studio (and its dependecies)
- MySQL (keep in Compose)
- Cloudant
```

lidderupk, 2020-06-17:
https://cloud.ibm.com/unifiedsupport/cases?number=CS1843744
- created ticket on 06/10
- updated with logs on 06/16

Contacted support on 06/16. They asked for the output of `IBMCLOUD_TRACE=true ibmcloud target --cf`. Sent them the following log for review on 06/16. Will contact for an update at the end of the week.

```
❯ IBMCLOUD_TRACE=true ibmcloud target --cf
❯ IBMCLOUD_TRACE=true ibmcloud target --cf

REQUEST: [2020-06-16T13:55:34-07:00]
GET /v2/regions HTTP/1.1
Host: mccp.us-south.cf.cloud.ibm.com
Accept: application/json
Accept-Language: en-US
Accept-Language: en
Authorization: [PRIVATE DATA HIDDEN]
Content-Type: application/json
User-Agent: IBM Cloud CLI 1.0.0 / darwin

RESPONSE: [2020-06-16T13:55:35-07:00] Elapsed: 473ms
HTTP/1.1 200 OK
Transfer-Encoding: chunked
Access-Control-Allow-Headers: Content-Type
Access-Control-Allow-Methods: GET,OPTIONS,POST,DELETE,PUT,PATCH
Access-Control-Allow-Origin: undefined
Connection: Keep-Alive
Content-Type: application/json; charset=utf-8
Date: Tue, 16 Jun 2020 20:55:34 GMT
Etag: W/"1087-vIsEK8beghJKFWMkBwvZwt/f6s8"
X-Backside-Transport: OK OK
X-Global-Transaction-Id: 63cd456d5ee931c6d358c183
X-Ratelimit-Limit: 150
X-Ratelimit-Remaining: 147
X-Ratelimit-Reset: 1592340934989

1087
[
  {
    "id": "ibm:yp:us-south",
    "domain": "ng.bluemix.net",
    "name": "us-south",
    "region": "us-south",
    "display_name": "Dallas",
    "customer": {
      "name": "ibm",
      "display_name": "IBM"
    },
    "deployment": {
      "name": "yp",
      "display_name": "Production"
    },
    "geo": {
      "name": "us-south",
      "display_name": "Dallas"
    },
    "public_regions_by_proximity": [
      "ibm:yp:us-south",
      "ibm:yp:us-east",
      "ibm:yp:eu-gb",
      "ibm:yp:eu-de",
      "ibm:yp:au-syd"
    ],
    "console_url": "https://console.bluemix.net",
    "cf_api": "https://api.us-south.cf.cloud.ibm.com",
    "mccp_api": "https://mccp.us-south.cf.cloud.ibm.com",
    "type": "public",
    "home": true,
    "aliases": [],
    "settings": {
[... 254 lines cut ...]
POST /cloudfoundry/login/us-south/oauth/token HTTP/1.1
Host: iam.cloud.ibm.com
Accept: application/json
Accept-Language: en-US
Accept-Language: en
Authorization: [PRIVATE DATA HIDDEN]
Content-Type: application/x-www-form-urlencoded
User-Agent: IBM Cloud CLI 1.0.0 / darwin

grant_type=iam_token&iam_token=[PRIVATE DATA HIDDEN]&response_type=uaa&scope=

RESPONSE: [2020-06-16T13:55:36-07:00] Elapsed: 231ms
HTTP/1.1 403 Forbidden
Connection: close
Content-Length: 734
Content-Language: en-US
Content-Type: application/json
Date: Tue, 16 Jun 2020 20:55:36 GMT
Set-Cookie: sessioncookie=a0e599d6f2c730f2f552a41c97434510; Path=/; Secure; HttpOnly
Transaction-Id: 90a99bb1e2b745afaf637d3221fee187

{"context":{"requestId":"90a99bb1e2b745afaf637d3221fee187","requestType":"incoming.Cf_OAuth_Token","userAgent":"IBM Cloud CLI 1.0.0 / darwin","url":"https://iam.cloud.ibm.com","instanceId":"iamid-5.6-8068-cbef4dc-76cb8f5f79-g5hww","threadId":"20f1df","host":"iamid-5.6-8068-cbef4dc-76cb8f5f79-g5hww","startTime":"16.06.2020 20:55:36:801 GMT","endTime":"01.01.1970 00:00:00:000 GMT","elapsedTime":"0","locale":"en_US","clusterName":"iam-id-prdal10-ljdb"},"errorCode":"BXNIM0204E","errorMessage":"No UAA user exists with the provided user name / IAM ID","errorDetails":"User name / IAM ID 'marco.rodriguez@prometeoplatform.com', environment info: region 'us-south'","error":"forbidden","error_description":"BMXLS0201E: No such account"}
FAILED
Could not login to Cloud Foundry: Remote server error. Status code: 403, error code: forbidden, message: BMXLS0201E: No such account
```

lidderupk, 2020-06-22:
Update from the support team on 06/22/2020

```
Hello,

My name is Anthony and I will be assisting you with this issue.

I have escalated this issue to our dev team, and am now standing by for a response from them. I attempted to push a fix for the issue, but the system hung up while trying to perform the fix, so we'll have to wait on dev for this one. Apologies.

Kind Regards,
Anthony S.
Revenue Support Enablement Team (RSET_

```

lidderupk, 2020-06-26:
The problem has been fixed by IBM support on 06/25/20. I am able to target cf using the command line and @mrodrise is now able to create compose database.

krook, 2020-07-09:
This is underway. Cloud credits increased.

krook, 2020-07-15:
Done!

closed 2020-07-15 by krook