Call-for-Code/Prometeo-Pyrrha #76: Redefine user stories: (1) Command Center, (2) Firefighter

krook, 2020-05-19:


krook, 2020-06-18:
@AlkAbr @samantadabis 

the materials we used for the workshop are here. This includes many photos of the activities and stickies: https://ibm.ent.box.com/folder/94069648510 

The hills, tasks, and roles were then entered into GitHub: https://github.com/Code-and-Response/Prometeo/wiki

krook, 2020-06-18:
This will involve re-interviewing @svaleroc and Joan about any changes needed.

samantadabis, 2020-06-18:
Thanks, @krook. In addition to reviewing the documents you sent, I met with Salome and Joan today. I am working on defining the key messages, audiences, and objectives, which will be a good starting point for when and how we develop communication plans.

krook, 2020-09-08:
I believe John has this documented along with Dina.