Call-for-Code/Prometeo-Pyrrha #84: Deep analysis about the 6 sensors to include in the IoT device: Carbon Monoxide, Nitrogen Dioxide, Acrolein, Formaldehyde, Benzene and GPS.

svaleroc, 2020-05-20:


krook, 2020-06-18:
Updated the description per the discussion today. The team is working on documentation and brainstorming in the Box folder: https://ibm.ent.box.com/folder/115859778363

krook, 2020-06-18:
Documenting the actual thresholds is a piece of work tracked in #86

mrodrise, 2020-06-22:
I uploaded the moto project proposal in /Hardware Design and Solution/Moto Project Proposal
They proposed ATMEGA2560 because with ESP32 there are no so much pins for everything. I think we should use ESP32 anyway but with less sensors and options.

For the GPS they propose to use SIM7000G because it has LTE-M and NB-IoT connectivity

For formaldehídos and benzeno exist MQ-138 and MQ-135 but they doesn't support more than 50 celsius degrees.

There is no sensor for including in a pcb for acroleina.

The other gas sensors (CO and nitrogen dioxide), they propose KEMET USEQGCCAC8L100.

The max operating temperature should be 80 celsius degrees.

krook, 2020-09-08:
We'll need to document these findings and prepare for a future iteration