Call-for-Code/Prometeo-Pyrrha #85: Solve the bug to start sending data by the device

svaleroc, 2020-05-20:


krook, 2020-05-20:
Simulator vs device issue. @derekteay to follow up with @joraco-dev

joraco-dev, 2020-05-20:
Please, @svaleroc @mrodrise could you ellaborate on this? I don't remember what was going here or what was the issue.

Thanks

mrodrise, 2020-06-18:
Hello, the problem here was that when you connect a device in order to send data to the IBM Cloud, doesn't appear the status of the firefighter in the dashboard.

There is a trick you can do, you can go to IOT platform simulation, simulate a device and then we'll appear both devices in the dashboard, the simulated and the real.

Regards,
Marco.

krook, 2020-06-18:
Ah, so a test from the simulator is required before any real data is shown. Does this happen just once for new devices, or is it a process that has to be worked around regularly?

mrodrise, 2020-06-18:
It happens regularly, I only test with one real device because I have only one. I think that if you have more than one device, there is not problem but I'm not sure. Maybe Josep can give more information.

closed 2020-09-06 by mrodrise