Call-for-Code/Prometeo-Pyrrha #89: Designer - Articulate Storyboards for phase 1

JSegrave-IBM, 2020-06-24:
Articulate Storyboards early on to enable firefighters, designers, software & hardware people review them before development and identify conflicts, issues and refinements (as well as have a shared sense of scope).

e.g. ’_Firefighter with a watch and smartphone receives ‘status red’ alert’_ - the storyboards can state things like:
* What the firefighter will see / hear / feel when alerted during field conditions (e.g. can take into account 8% of men are red-green color-blind).
* Where the visible things are and what they're doing - visibility of the sensor LED / Smart Watch / Phone under field conditions.
* What happens when 1 device (sensor) is in play - how the alert goes off, how a firefighter responds to or cancels it.
* What happens when 1 combinations of devices are in play (sensor / phone / watch). How a firefighter responds to or cancels these. Which device is mainly responsible for alerts? (would multiple devices alerting be OK / confusing?) etc...

JSegrave-IBM, 2020-06-24:
Note: in order to work out the tech (algos / ML) behind the alerts, we're going to need to do some of this storyboarding.

e.g. (just one illustrative example)
In the command center, an alert goes red for Alfonso. What happens next?
* Does Alfonso's response depend on knowing what the underlying condition, or conditions were? e.g. will he react differently if
** a sensor just saw a toxic level of NO2 for a few seconds and then stopped
**  a firefighter is experiencing a combination of {high temp + CO}?
**  a firefighter has been exposed to too much CO cumulatively over the last 15 minutes?

Labelling data is one of the big costs in machine learning and the 'type' of explainability required determines how we do the labels (as well as how feedback is gathered at runtime.) e.g. red / yellow /green is simple, but is it sufficiently-explained to enable correct follow-up actions?  When context/explanation is essential, we often choose to use machine learning to learn the *explanations* (like _'critical exposure to NO2 over 15 mins'_) rather than the _decisions_ (_'red: get the firefighter out'_).  Need to know this up-front, as labelling 100s/1000s of examples is expensive is usually not cost-effective to repeat / fix.

krook, 2020-09-08:
This should be updated for our October 1 MVP.