Call-for-Code/Prometeo-Pyrrha #90: Developer: Explore mobile app framework and SDKs

krook, 2020-06-24:
We need to determine whether we go straight into an Android platform and mobile device manufacturer SDK or if we start off generic and more web app focused.

Need to evaluate the options

- [ ] Android (Kotlin/Java), native SDKs needed (push notifications, phone/watch, heart rate)
- [ ] React Native (JavaScript) to build to Java and Swift automatically. Native APIs exposed through plugins.
- [ ] Progressive web app using the device browser. Has some features like GPS and push notifications, but maybe not all

lidderupk, 2020-06-26:
The devices identified in the box note are:

- Galazy Xcover Pro. Smartphone Rugerizado: https://www.samsung.com/es/business/smartphones/galaxy-xcover-pro-g715/
- Galaxy Watch: https://www.samsung.com/es/wearables/galaxy-watch-r800/

Notes from Research:
**Galazy Xcover Pro.**
- Android 10 and Samsung One UI 2.0
- can use Java or Kotlin. What is the team skillset?
- native pros
   - easier to follow OS design principles
   - easier to do animations and more complex layouts
   - well established IDE/build tools
   - look and feel more native
- react-native pros
   - fast build time
   - (mostly) same code base for android/iOS
   - JS is common

**Galaxy Watch**
- Uses Tizen OS
   - web development (js, html)
   - .net development with C# (recommended for UI apps)
   - native development on Tizen OS (C++)
- Links
   - https://developer.samsung.com/sdp/blog/en-us/2020/06/23/tizen-tidbits-working-with-sensors-and-the-heart-rate-monitor
   - Sample code - https://developer.samsung.com/tizen/Galaxy-Watch/Resources/Samples/Galaxy-Watch.html
   - https://developer.samsung.com/sdp/blog/en-us/2019/10/02/use-tizen-web-to-measure-heart-rate-with-galaxy-watches

krook, 2020-07-09:
We decided here to:
- Use the native Java Android environment for the phone
- Web development on the watch

@lidderupk @narengan can you confirm the watch decision?

krook, 2020-09-08:
I've confirmed the watch decision.

closed 2020-09-08 by krook