Call-for-Code/Prometeo-Pyrrha #106: Draft design storyboard / walkthrough of Prometeo vision

JSegrave-IBM, 2020-07-31:
As a Prometeo contributor, I need to be walked-through a storyboard of the Prometeo Platform being used by a set of Firefighters and a Control Center Leader, so that I can be confident I've understood the objectives and considered what tasks need to be on plan in order to meet those objectives (particularly those related to my role).

This will be an iterative process - the goal is to get a rough draft up quickly to help everyone ask good questions and spot issues. Then refine in a series of iterations from there. "Perfect is the enemy of good enough".  We need just enough of a look-ahead to mitigate the risk of 'throw-away' work (e.g. if getting 7+ risk factors on screen in the end would need a very different design than the initial 3, then it's helpful to know that before we do the 3).

There will be lots of Questions to consider - e.g.
* What is it the Command Center leader needs to know?  'Now' vs 'cumulative' vs 'exposure' vs 'limits'? (Line-Series chart or simple 10 min / 30 min / 60 min / 4 hr / 8 hr Cumulative exposure summary, as per AEGLs?)
* Green/red/yellow need defined meanings. Propose that 'yellow' always means 'predicted to turn red within N minutes"? And Red means 'over the threshold configured by the client'
* Once it goes Yellow or Red, how long does it stay there?
* How to represent for multiple risks on screen simultaneously? Do FF just 'go red', or do we see red on the top row, yellow on the next row, etc?  How to make it easy to focus in the right place?
* What happens when there are more firefighters than screen space? Need to keep all critical info on the screen, not hidden on a second screen.
* How much detail? None / some? Only see detail when there's a problem, or when I specifically look?
* Registration - In an event with many IOT Devices, how does the FF know their device is registered to their phone? Do the devices have IDs? Don't want FF data getting mixed up.

krook, 2020-09-08:
@JSegrave-IBM do we have answers to these questions?

Adityamerc, 2021-03-13:
Can I be of any assistance in this too? I'm a full time UI dev proficient is JS and React.