Call-for-Code/Prometeo-Pyrrha #112: Define schema for sensor data table

JSegrave-IBM, 2020-09-09:
Proposed Schema below.

Note: suggest using the gas name, because while 'CO' and 'NO2' have become familiar, CH2O, C3H4O and C6H3(CH3)3 ... not so much :-)  not a strong opinion though (it's a database, only technical people will see it!), so feel free to change this.

| Key | TYPE | Name | Notes / Questions |
|:--- |:------|:-----|:-----|
| YES | STRING | SensorID | Should this be 'FirefighterID' now? |
| YES | DATETIME | timestamp | Important that these keys be minute-quantized - e.g. '10/02/2020 09:40' not '10/02/2020 09:40:34'
| | FLOAT | **temperature** | |
| | FLOAT | **humidity** | |
| | FLOAT | **carbon_monoxide** | |
| | FLOAT | carbon_monoxide_twa_10min | |
| | FLOAT | carbon_monoxide_twa_30min | |
| | FLOAT | carbon_monoxide_twa_60min | |
| | FLOAT | carbon_monoxide_twa_4hr | |
| | FLOAT | carbon_monoxide_twa_8hr | |
| | INT | carbon_monoxide_gauge_10min | |
| | INT | carbon_monoxide_gauge_30min | |
| | INT | carbon_monoxide_gauge_60min | |
| | INT | carbon_monoxide_gauge_4hr | |
| | INT | carbon_monoxide_gauge_8hr | |
| | FLOAT | **nitrogen_dioxide** | |
| | FLOAT | nitrogen_dioxide_twa_10min | |
| | FLOAT | nitrogen_dioxide_twa_30min | |
| | FLOAT | nitrogen_dioxide_twa_60min | |
| | FLOAT | nitrogen_dioxide_twa_4hr | |
| | FLOAT | nitrogen_dioxide_twa_8hr | |
| | INT | nitrogen_dioxide_gauge_10min | |
| | INT | nitrogen_dioxide_gauge_30min | |
| | INT | nitrogen_dioxide_gauge_60min | |
| | INT | nitrogen_dioxide_gauge_4hr | |
| | INT | nitrogen_dioxide_gauge_8hr | |
| | FLOAT | **formaldehyde** | |
| | FLOAT | formaldehyde_twa_10min | |
| | FLOAT | formaldehyde_twa_30min | |
| | FLOAT | formaldehyde_twa_60min | |
| | FLOAT | formaldehyde_twa_4hr | |
| | FLOAT | formaldehyde_twa_8hr | |
| | INT | formaldehyde_gauge_10min | |
| | INT | formaldehyde_gauge_30min | |
| | INT | formaldehyde_gauge_60min | |
| | INT | formaldehyde_gauge_4hr | |
| | INT | formaldehyde_gauge_8hr | |
| | FLOAT | **acrolein** | |
| | FLOAT | acrolein_twa_10min | |
| | FLOAT | acrolein_twa_30min | |
| | FLOAT | acrolein_twa_60min | |
| | FLOAT | acrolein_twa_4hr | |
| | FLOAT | acrolein_twa_8hr | |
| | INT | acrolein_gauge_10min | |
| | INT | acrolein_gauge_30min | |
| | INT | acrolein_gauge_60min | |
| | INT | acrolein_gauge_4hr | |
| | INT | acrolein_gauge_8hr | |
| | FLOAT | **benzene** | |
| | FLOAT | benzene_twa_10min | |
| | FLOAT | benzene_twa_30min | |
| | FLOAT | benzene_twa_60min | |
| | FLOAT | benzene_twa_4hr | |
| | FLOAT | benzene_twa_8hr | |
| | INT | benzene_gauge_10min | |
| | INT | benzene_gauge_30min | |
| | INT | benzene_gauge_60min | |
| | INT | benzene_gauge_4hr | |
| | INT | benzene_gauge_8hr | |

JSegrave-IBM, 2020-09-10:
@lidderupk @mrodrise - how's something like this for a table definition? Key points:
* Key is ('timestamp', 'firefighter_id')
* temp, humidity and all gauge percentages are [SMALLINT](https://mariadb.com/kb/en/smallint/)
* sensors and all time-weighted averages - are [FLOAT](https://mariadb.com/kb/en/float/) 
* I notice in the MariaDB database dump that all the create statements have DEFAULT CHARSET defaulted to 'latin1'. That might be OK for this table (no screen-visible values, just numbers, dates and IDs), but I thought I'd mention because there are other tables that should cater for culturally-diverse text, right? (e.g. firefighter names). Should this it be defaulting to something like UTF-8 instead?

CREATE TABLE 'metrics_dev' (
  'timestamp' timestamp NOT NULL,
  'firefighter_id' varchar(20) NOT NULL,
  'temperature' SMALLINT DEFAULT NULL,
  'humidity' SMALLINT DEFAULT NULL,
  'carbon_monoxide' FLOAT DEFAULT NULL,
  'carbon_monoxide_twa_10min' FLOAT DEFAULT NULL,
  'carbon_monoxide_twa_30min' FLOAT DEFAULT NULL,
  'carbon_monoxide_twa_60min' FLOAT DEFAULT NULL,
  'carbon_monoxide_twa_4hr' FLOAT DEFAULT NULL,
  'carbon_monoxide_twa_8hr' FLOAT DEFAULT NULL,
  'carbon_monoxide_gauge_10min' SMALLINT DEFAULT NULL,
  'carbon_monoxide_gauge_30min' SMALLINT DEFAULT NULL,
  'carbon_monoxide_gauge_60min' SMALLINT DEFAULT NULL,
  'carbon_monoxide_gauge_4hr' SMALLINT DEFAULT NULL,
  'carbon_monoxide_gauge_8hr' SMALLINT DEFAULT NULL,
  'nitrogen_dioxide' FLOAT DEFAULT NULL,
  'nitrogen_dioxide_twa_10min' FLOAT DEFAULT NULL,
  'nitrogen_dioxide_twa_30min' FLOAT DEFAULT NULL,
  'nitrogen_dioxide_twa_60min' FLOAT DEFAULT NULL,
  'nitrogen_dioxide_twa_4hr' FLOAT DEFAULT NULL,
  'nitrogen_dioxide_twa_8hr' FLOAT DEFAULT NULL,
  'nitrogen_dioxide_gauge_10min' SMALLINT DEFAULT NULL,
  'nitrogen_dioxide_gauge_30min' SMALLINT DEFAULT NULL,
  'nitrogen_dioxide_gauge_60min' SMALLINT DEFAULT NULL,
  'nitrogen_dioxide_gauge_4hr' SMALLINT DEFAULT NULL,
  'nitrogen_dioxide_gauge_8hr' SMALLINT DEFAULT NULL,
  'formaldehyde' FLOAT DEFAULT NULL,
  'formaldehyde_twa_10min' FLOAT DEFAULT NULL,
  'formaldehyde_twa_30min' FLOAT DEFAULT NULL,
  'formaldehyde_twa_60min' FLOAT DEFAULT NULL,
  'formaldehyde_twa_4hr' FLOAT DEFAULT NULL,
  'formaldehyde_twa_8hr' FLOAT DEFAULT NULL,
  'formaldehyde_gauge_10min' SMALLINT DEFAULT NULL,
  'formaldehyde_gauge_30min' SMALLINT DEFAULT NULL,
  'formaldehyde_gauge_60min' SMALLINT DEFAULT NULL,
  'formaldehyde_gauge_4hr' SMALLINT DEFAULT NULL,
  'formaldehyde_gauge_8hr' SMALLINT DEFAULT NULL,
  'acrolein' FLOAT DEFAULT NULL,
  'acrolein_twa_10min' FLOAT DEFAULT NULL,
  'acrolein_twa_30min' FLOAT DEFAULT NULL,
  'acrolein_twa_60min' FLOAT DEFAULT NULL,
  'acrolein_twa_4hr' FLOAT DEFAULT NULL,
  'acrolein_twa_8hr' FLOAT DEFAULT NULL,
  'acrolein_gauge_10min' SMALLINT DEFAULT NULL,
  'acrolein_gauge_30min' SMALLINT DEFAULT NULL,
  'acrolein_gauge_60min' SMALLINT DEFAULT NULL,
  'acrolein_gauge_4hr' SMALLINT DEFAULT NULL,
  'acrolein_gauge_8hr' SMALLINT DEFAULT NULL,
  'benzene' FLOAT DEFAULT NULL,
  'benzene_twa_10min' FLOAT DEFAULT NULL,
  'benzene_twa_30min' FLOAT DEFAULT NULL,
  'benzene_twa_60min' FLOAT DEFAULT NULL,
  'benzene_twa_4hr' FLOAT DEFAULT NULL,
  'benzene_twa_8hr' FLOAT DEFAULT NULL,
  'benzene_gauge_10min' SMALLINT DEFAULT NULL,
  'benzene_gauge_30min' SMALLINT DEFAULT NULL,
  'benzene_gauge_60min' SMALLINT DEFAULT NULL,
  'benzene_gauge_4hr' SMALLINT DEFAULT NULL,
  'benzene_gauge_8hr' SMALLINT DEFAULT NULL,
  PRIMARY KEY ('timestamp', 'firefighter_id')
) ENGINE=InnoDB DEFAULT CHARSET=latin1;

krook, 2020-09-10:
Thanks @JSegrave-IBM!

Yes, the database and each table should be UTF-8. We may have to dump data and recreate it:

```sql
CREATE DATABASE IF NOT EXISTS prometeo character set UTF8mb4 collate utf8mb4_unicode_ci;
```

mrodrise, 2020-09-10:
Thanks for all, great work.

I consider that we should add a field for the device id that is sending the data

Maybe we can add fields for the status color that gives the algorithm in that moment, only to have the history

We can change the name of the table... I never liked metrics... we can rename to firefighters_status_readings... or whatever...

JSegrave-IBM, 2020-09-11:
Thanks @mrodrise - table name is now 'firefighter_status_analytics' - is that OK? (mix of calculated values and 'cleaned' sensor values - e.g. dropout-refilled, minute-aligned ). Also added 'device_id' and 'device_battery_level' so all the screen display data is in one place.

JSegrave-IBM, 2020-09-11:
> ```sql
> CREATE DATABASE IF NOT EXISTS prometeo character set UTF8mb4 collate utf8mb4_unicode_ci;
> ```

Thanks @krook - I presume all the CREATE TABLE statements should have DEFAULT CHARSET=UTF8mb4; in kind?

krook, 2020-09-11:
@JSegrave-IBM correct. But it may automatically cascade as the default. Looking at another database I have:

[image]

JSegrave-IBM, 2020-09-17:
**Changes:** (FYI -  @mrodrise @lidderupk )

* There are two tables:
  * `firefighter_sensor_log` where each row is the raw/unmodified sensor readings associated with `firefighter_id` during the minute `timestamp_mins` - we never modify this data
  * `firefighter_status_analytics` - where each row the same data as the sensor log, plus all the analytic information about `firefighter_id` during the minute `timestamp_mins`. This is a kind of one-stop-shop for data analysis - everything about the  firefighter in one place.  Making it a separate table to `firefighter_sensor_log` allows us to keep an untouched record of what was sent. It also removes a few potential contention points between the processes that read and write information.
* Added `device_timestamp` - the unmodified original device timestamp (not minute-quantized)

JSegrave-IBM, 2020-09-18:
Normally the `device_status_LED` and `analytics_status_LED` (i.e. the LED colors) will be the same, but in a disconnected scenario, they may be different. We'll capture both. Hence these two updates above:
* `firefighter_sensor_log` - added `device_status_LED`
* `firefighter_status_analytics` - added `device_status_LED` and `analytics_status_LED`

JSegrave-IBM, 2020-09-22:
Pull request: https://github.com/Code-and-Response/Prometeo-Database/pull/3

JSegrave-IBM, 2020-10-02:
Modified DB field names to make them less sensitive to configuration changes. Fields for the different time-window TWAs are now named the after the number of minutes in that window (stable) and not by the display name (less stable and prone to SQL issues).